silencht/SG-SLAM #1: rviz has nothing to show and octomap tree is empty

ahhhname, 2023-04-10:
Thanks for your work. when I reproduct it,following the step in the readme, I finished all the steps for  Building SG-SLAM successfully , but  when I run and test, it occur some problems.
roslaunch octomap.launch show that:
[image]

 [WARN] [1681106496.736057330]: Nothing to publish, octree is empty
and Rviz has nothing to show.

the test instrutions are the follows from Readme
###
roscore
cd SG-SLAM/src/octomap_server/launch
roslaunch octomap.launch
roslaunch transform.launch
rviz
cd SG-SLAM/src/sg-slam/
./run_tum_walking_xyz.sh
###

silencht, 2023-04-10:
Have you subscribed to related topics in Rviz? For example /occupied_cells_vis_array, /SG_SLAM/Semantic_Objects.
Or you can use the rviz configuration file to configure directly (its path is located in SG-SLAM/src/sg-slam/Examples/rvizconfig.rviz).

If it is determined that the above steps have been practiced, you can troubleshoot the problem step by step from the topic posting.

If you have more questions, please continue to reply in issues or send an email directly (silencht@qq.com)

ahhhname, 2023-04-10:
Thank u. I should run " rviz -d path/rvizconfig.rviz" to make the .rviz file have the effect instead of run "rviz" directly. I advice you can have a prompt in the readme. may be just because I'm newer to this. but the result doesn't look perfect like the picture in the readme.
[image]
the map are the octree map, not the object metric map like the below:
[image]
And in the left menu, show the semantic object "Uninitialized quaternion, assuming identity."
[image]
and the image don't show up. could you give some help me to fix it?

silencht, 2023-04-10:
Thanks for the suggestion, the README section will be revised later.

Warning information can be ignored temporarily.

The picture you mentioned is the result of manually linking the image frame and the map with arrows, which is not the actual effect. Additional map images have been added to the README for reference.

Later, some notes about empirical parameters in the program and configuration files will be updated.

40526136, 2023-06-10:
Thanks for your work. when I reproduct it,following the step in the readme, I finished all the steps for Building SG-SLAM successfully , but when I run and test, it occur some problems.
roslaunch octomap.launch show that:
[image]

40526136, 2023-06-10:
implement：rviz -d atcatkin_ws/src/SG-SLAM/src/sg-slam/Examples/rvizconfig.rviz
[image]

silencht, 2023-06-10:
octomap.launch should be fine, I think the problem may be that rviz subscribes to the relevant nodes correctly.

rviz -d catkin_ws/src/SG-SLAM/src/sg-slam/Examples/rvizconfig.rviz

This command uses the rvizconfig.rviz file to configure and start rviz. You should double check that the path is correct. From the picture you sent, it seems that rviz has not successfully subscribed to the relevant nodes.

40526136, 2023-06-10:
> octomap.launch應該沒問題，我覺得問題可能是rviz正確訂閱了相關節點。
> 
> rviz -d catkin_ws/src/SG-SLAM/src/sg-slam/Examples/rvizconfig.rviz
> 
> 此命令使用 rvizconfig.rviz 文件來配置和啟動 rviz。您應該仔細檢查路徑是否正確。從你發的圖片來看，rviz好像還沒有成功訂閱相關節點。

Thank you for your opinion

silencht, 2023-06-10:
You can also click to load the configuration file "rvizconfig.rviz" from the rviz GUI button file. Alternatively, you can manually add related nodes.

40526136, 2023-06-10:
hello author
I would like to ask the author if SG-SLAM can run the camera in the physical environment

silencht, 2023-06-10:
The astro pro RGB-D camera I used was tested in a real environment, and part of the video in readme.md was recorded in this situation. Some parameters may need to be adjusted according to the actual scene, and the functions of these parameters are generally explained in readme.md.

40526136, 2023-06-10:
hello author
I changed the path but still the screen is not displayed
[image]

silencht, 2023-06-10:
After you run ./run_tum_walking_xyz.sh in the picture, it will generally display the initialization information of ORB-SLAM2 and some configuration information of SG-SLAM. So you can check if the command path in the run_tum_walking_xyz.sh file is correct.

40526136, 2023-06-10:
hello author
What about the rviz problem?
[image]

silencht, 2023-06-10:
I think there is nothing wrong with rivz now, the problem is that SG-SLAM is not running successfully. Two image topic subscriptions are set at runtime in a real scene using an RGB-D camera. They do not work if using the tum dataset.

40526136, 2023-06-10:
hello author
Below is my path '.
[image]
/run_tum_walking_xyz.sh'

silencht, 2023-06-10:
The path should be wrong. You can refer to ORB-SLAM2 to reset these paths so that SG-SLAM can run correctly. If there is a problem with the relative path, you can use the absolute path first.

40526136, 2023-06-11:
hello author
This is the problem I'm currently having
[image]

silencht, 2023-06-12:
Sorry. I haven't encountered this kind of problem so far, so I can't directly give a solution. Maybe you can search for similar information on related webpages to try to solve it.

40526136, 2023-06-12:
hello author
I can solve some of the problems, but there are still some small problems that I can't solve.
[image]

There is still a problem with MarkerArray on rviz

silencht, 2023-06-12:
Despite these warnings, the map bulit in the picture does not appear to have errors? It may be a problem with the [TF and TF2 systems](http://wiki.ros.org/tf2/Migration) of ROS, but it didn't happen when I used it.

40526136, 2023-06-12:
> 可能是ROS的TF和TF2系統
Where can this question be amended?

silencht, 2023-06-12:
I haven't had this problem so far, so I can't reproduce it. You can check some related web pages or forums to try to solve it.

40526136, 2023-06-12:
> 到目前為止我還沒有遇到過這個問題，所以我無法重現它。可以查看一些相關的網頁或者論壇來嘗試解決。

hello author
What about this problem?
Building a map on rviz does not preserve the surrounding environment

silencht, 2023-06-12:
It may be that the parameters at the end of the .yaml file are not set to the mode you want, you can read README.md to understand them. 

In addition, one of the two point cloud topics subscribed by rviz is the point cloud display of the current key frame, and the other is the global 3D point cloud reconstruction map display. I guess, what you may be talking about is the effect of the first point cloud topic.

40526136, 2023-06-12:
> It may be that the parameters at the end of the .yaml file are not set to the mode you want, you can read README.md to understand them.
>
> In addition, one of the two point cloud topics subscribed by [...]

I subscribe to /octomap_point_cloud_centers showing red
[image]

40526136, 2023-06-12:
hello author
Rviz tf error: Failed to transform from frame [/map] to frame [map]
It seems to be the relationship between ubuntu20 and ubuntu18
I would like to ask the author is from ubuntu18, where is the frame_id to "/map" in the author's code？

40526136, 2023-06-13:
> Related locations related to "/map" are:
> 
> 1. https://github.com/silencht/SG-SLAM/blob/69f5a0635f501fe60803cb1828924d486ae37717/src/octomap_server/launch/octomap.launch#L13
> 2. https://github.com/silencht/SG-SLAM/blob/69f5a0635f501fe60803cb1828924d486ae37717/src/sg-slam/Examples/rgbd_tum.cc#L207
> 3. https://github.com/silencht/SG-SLAM/blob/69f5a0635f501fe60803cb1828924d486ae37717/src/sg-slam/Examples/ros_rgbd.cc#L160
> 4. https://github.com/silencht/SG-SLAM/blob/69f5a0635f501fe60803cb1828924d486ae37717/src/sg-slam/src/PointcloudMapping.cc#L284
> 5. https://github.com/silencht/SG-SLAM/blob/69f5a0635f501fe60803cb1828924d486ae37717/src/sg-slam/src/PointcloudMapping.cc#L352
> 6. https://github.com/silencht/SG-SLAM/blob/69f5a0635f501fe60803cb1828924d486ae37717/src/sg-slam/src/PointcloudMapping.cc#L387
> 7. https://github.com/silencht/SG-SLAM/blob/69f5a0635f501fe60803cb1828924d486ae37717/src/sg-slam/src/PointcloudMapping.cc#L404

Thank you author for being successful

closed 2023-06-17 by silencht

QQQ300kuai, 2023-12-18:
I also encountered the same problem, can you tell me how to solve it?

jiatianluo, 2026-01-18:
您好，我也遇到了类似的问题，在运行./run_tum_walking_xyz.sh命令后，终端显示成功运行，并且

[image]

但是我运行octomap.launch 时，终端显示[WARN] [1768727223.243441538]: Nothing to publish, octree is empty
，并且运行rviz -d catkin_ws/src/SG-SLAM/src/sg-slam/Examples/rvizconfig.rviz命令后，rviz什么画面都没有，也无任何报错。请问您可以提供一些指导意见吗