silencht/SG-SLAM #4: 运行 ./run_tum_walking_xyz.sh 时没有反应

Justdo1t-lzw, 2023-05-19:
代码编译通过，运行 ./run_tum_walking_xyz.sh 时没有反应
[image]
roslaunch octomap.launch 的时候显示： Nothing to publish, octree is empty
[image]
最新版本的代码中没有 transform.launch 从之前版本找的
rviz中也没有显示，是使用的 rvizconfig.rviz
这是我的topic
[image]
求教有没有什么解决办法，感谢

silencht, 2023-05-19:
新版本代码中不需要transform.launch，检查./run_tum_walking_xyz.sh命令的路径，对应路径是否有相关文件及数据集。或者直接手动输入命令测试。topic看起来没有问题。

Justdo1t-lzw, 2023-05-20:
感谢您的帮助，我已经成功运行示例，我的错误是tum数据集没有使用associate.py同步，修改后运行效果看起来不错，感谢您的工作。

closed 2023-05-20 by silencht

MR-DKQ, 2024-07-20:
> associate.py

出现了一个./Examples/sg_slam_tum: not found错误，感觉前面的编译没有错误，可能是哪方面出现了问题呢