silencht/SG-SLAM #5: What are target detection？

40526136, 2023-06-11:
hello author
Can I ask what objects can be detected

silencht, 2023-06-12:
Since the mobilenetV3-SSD model that has been trained is used, there are 20 types of objects that can be detected, which are : [object class that can be detected](https://github.com/silencht/SG-SLAM/blob/69f5a0635f501fe60803cb1828924d486ae37717/src/sg-slam/src/Detector2D.cc#LL8C1-L14C7)

The model above comes from : https://github.com/Tencent/ncnn/blob/master/examples/mobilenetv3ssdlite.cpp

You can also use other examples of ncnn or train the network model yourself.

40526136, 2023-06-14:
hello author
After detecting the screen and chair objects, and in the semantic segmentation, the red and blue three-dimensional boxes are displayed on rviz, but the people and potted plants are not displayed on rviz. Where can these programs be written.

silencht, 2023-06-17:
Most of the code is located in Detector2D.cc, Detector3D.cc and [PointcloudMapping.cc](https://github.com/silencht/SG-SLAM/blob/main/src/sg-slam/src/PointcloudMapping.cc) in src/sg-slam/src. It should be noted that, as a highly dynamic object, people have been eliminated in advance when building the map.

closed 2023-08-28 by silencht