silencht/SG-SLAM #7: Problems encountered in running fr3_walking other datasets

wahaahaaaa, 2023-08-24:
Hello author, when I run fr3_walking_xyz, when the camera deflects to the left, there is a situation where there are no feature points;
At the same time, when several other data sets are running, there is no running response and no window appears;
May I ask how to solve it, is it because of my computer configuration?

silencht, 2023-08-28:
No feature points should be lost in the system track. There are several possible reasons for this, try adjusting the following parameter settings in tum3.yaml. See the README for a description of the parameters.

The appearance of no windows requires more information from you.

wahaahaaaa, 2023-08-28:
TUM3.yaml文件完全使用的是作者你所给出的参数，运行所出现的问题我已录制，视频链接如下【fr3walking_xyz消失，track lost】 https://www.bilibili.com/video/BV1Uu4y1D7jJ/?share_source=copy_web&vd_source=1492107fa7aea466b4ba721f532497fe，问题出现在第20秒，track lost，望回复，感谢不尽！

silencht, 2023-08-28:
Contact information has been provided in the video comments

wahaahaaaa, 2023-08-28:
Did not find your contact information in the video comment, please provide it again, thanks！

closed 2023-09-14 by silencht