silencht/SG-SLAM #9: 关于使用相机运行

qqqaah, 2023-09-14:
作者你好，您提到可以用硬件相机运行程序，但是我使用astrapro相机运行程序时，会直接闪退，并且显示段错误，请问您能提供点用相机运行的细节吗？期待回复，非常感谢！

silencht, 2023-09-14:
针对相机硬件，修改.yaml内的相机内参了吗？

qqqaah, 2023-09-14:
我使用了能够在ORB-SLAM2中运行的.yaml文件的内参，即使如此，也是运行就闪退，但是在运行ORB-SLAM2原版代码时没有问题

silencht, 2023-09-14:
简单流程：
1. 使用相机输出两个ROS话题（RGB图像流+深度图像流，话题名称见下面代码）
2. 根据相机硬件的参数，修改src/sg-slam/Examples/astra_pro_camera.yaml文件内的相机内参等信息
3. 使用src/sg-slam/run_astra_pro_camera.sh启动系统后，系统会订阅ROS系统的图像流话题。话题订阅代码位于src/sg-slam/Examples/ros_rgbd.cc
```c++
message_filters::Subscriber<sensor_msgs::Image> rgb_sub(nh, "/camera/rgb/image_raw", 30);
message_filters::Subscriber<sensor_msgs::Image> depth_sub(nh, "/camera/depth_registered/image_raw", 30);
```

silencht, 2023-09-14:
使用gdb测试代码闪退的位置
```shell
cd SG-SLAM/src/sg-slam
gdb ./Examples/sg_slam_ros_rgbd
#进入gdb后
run ./Vocabulary/ORBvoc.bin ./Examples/astra_pro_camera.yaml
```
查看闪退时的代码定位

qqqaah, 2023-09-14:
gdb ./Examples/sg_slam_ros_rgbd 
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./Examples/sg_slam_ros_rgbd...(no debugging symbols found)...done.
(gdb) run ./Vocabulary/ORBvoc.bin ./Examples/astra_pro_camera.yaml
Starting program: /home/ayuyu/SG-SLAM/src/sg-slam/Examples/sg_slam_ros_rgbd ./Vocabulary/ORBvoc.bin ./Examples/astra_pro_camera.yaml
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffb100d700 (LWP 6376)]
[New Thread 0x7fffb080c700 (LWP 6377)]
[New Thread 0x7fffae00b700 (LWP 6378)]
[New Thread 0x7fffa980a700 (LWP 6379)]
[New Thread 0x7fffa9009700 (LWP 6380)]
[New Thread 0x7fffa6808700 (LWP 6381)]
[New Thread 0x7fffa2007700 (LWP 6382)]
[New Thread 0x7fffa1806700 (LWP 6383)]
[New Thread 0x7fff9f005700 (LWP 6384)]
[New Thread 0x7fff9c804700 (LWP 6385)]
[New Thread 0x7fff98003700 (LWP 6386)]
[New Thread 0x7fff95802700 (LWP 6387)]
[New Thread 0x7fff95001700 (LWP 6388)]
[New Thread 0x7fff90800700 (LWP 6389)]
[New Thread 0x7fff8dfff700 (LWP 6390)]
[New Thread 0x7fff8b59f700 (LWP 6392)]
[New Thread 0x7fff8ad9e700 (LWP 6393)]
[New Thread 0x7fff8a59d700 (LWP 6394)]
[New Thread 0x7fff89d9c700 (LWP 6395)]

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

is map construction consider dynamic: 0
is global pointcloud reconstruction: 0
is octomap semanticObject map construction: 1
camera_valid_depth_Min is 0.5
camera_valid_depth_Max is 5
Sor_Local_MeanK is 50
Sor_Local_StddevMulThresh is 2
Voxel_Local_LeafSize is 0.01
Sor_Global_MeanK is 50
Sor_Global_StddevMulThresh is 2
Voxel_Global_LeafSize is 0.01
Detector3D_Sor_MeanK is 50
Detector3D_Sor_StddevMulThresh is 1
Detector3D_Voxel_LeafSize is 0.01
Detect3D_EuclideanClusterTolerance is 0.02
Detect3D_EuclideanClusterMinSize is 1000
Detect3D_EuclideanClusterMaxSize is 30000
Detect3D_DetectSimilarCompareRatio is 0.1
global_pc_update_kf_threshold is 30
[New Thread 0x7fff894b9700 (LWP 6396)]
Start PointCloudMapping Viewer...

Camera Parameters: 
- fx: 577.547
- fy: 578.633
- cx: 310.243
- cy: 253.655
- k1: 0.10105
- k2: -0.160047
- p1: 0.002287
- p2: 0.000271
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 2.77034
[New Thread 0x7fff88cb8700 (LWP 6397)]
[New Thread 0x7fff67fff700 (LWP 6398)]

detection_confidence_threshold is 0.99

dynamic_detection_confidence_threshold is 0.05
[New Thread 0x7fff6770d700 (LWP 6399)]
[New Thread 0x7fff66f0c700 (LWP 6400)]
[New Thread 0x7fff6670b700 (LWP 6401)]
[New Thread 0x7fff65dcf700 (LWP 6402)]
Detector2D Thread start ...
[New Thread 0x7fff655ce700 (LWP 6403)]
[New Thread 0x7fff58c25700 (LWP 6404)]
[New Thread 0x7fff53fff700 (LWP 6405)]
[New Thread 0x7fff537fe700 (LWP 6406)]
[New Thread 0x7fff52ffd700 (LWP 6407)]
[New Thread 0x7fff527fc700 (LWP 6408)]
[New Thread 0x7fff51ffb700 (LWP 6409)]
[New Thread 0x7fff517fa700 (LWP 6410)]
[New Thread 0x7fff50ff9700 (LWP 6411)]
[New Thread 0x7fff33fff700 (LWP 6412)]
[New Thread 0x7fff337fe700 (LWP 6413)]
[New Thread 0x7fff32ffd700 (LWP 6414)]
[New Thread 0x7fff327fc700 (LWP 6415)]
[New Thread 0x7fff31ffb700 (LWP 6417)]
[New Thread 0x7fff317fa700 (LWP 6418)]
[New Thread 0x7fff30ff9700 (LWP 6419)]
[New Thread 0x7fff13fff700 (LWP 6420)]
[New Thread 0x7fff137fe700 (LWP 6421)]
[New Thread 0x7fff12ffd700 (LWP 6422)]
[New Thread 0x7fff127fc700 (LWP 6423)]
[New Thread 0x7fff11ffb700 (LWP 6424)]
[New Thread 0x7fff117fa700 (LWP 6425)]
[New Thread 0x7fff10ff9700 (LWP 6426)]
[New Thread 0x7ffef3fff700 (LWP 6427)]
[New Thread 0x7ffef37fe700 (LWP 6428)]
[New Thread 0x7ffef2ffd700 (LWP 6429)]
[New Thread 0x7ffef27fc700 (LWP 6430)]
[New Thread 0x7ffef1ffb700 (LWP 6431)]
[New Thread 0x7ffef17fa700 (LWP 6432)]
[New Thread 0x7ffef0ff9700 (LWP 6433)]
[New Thread 0x7ffed3fff700 (LWP 6434)]
[New Thread 0x7ffed37fe700 (LWP 6435)]
[New Thread 0x7ffed2ffd700 (LWP 6436)]
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.[New Thread 0x7ffed27fc700 (LWP 6437)]
[New Thread 0x7ffed1ffb700 (LWP 6438)]

Thread 1 "sg_slam_ros_rgb" received signal SIGSEGV, Segmentation fault.
0x00007ffff57ab01b in ros::CallbackQueue::callAvailable(ros::WallDuration)
    () from /opt/ros/melodic/lib/libroscpp.so
(gdb) 
这些就是gdb的调试信息，问题应该是最后那里，但是我不知道这个问题是代码的哪个位置

silencht, 2023-09-14:
emmm，从报错信息来看是ROS的队列回调函数可能有问题，我也没遇到过，你可以检查一下：
1. 相机发布图像话题和订阅话题完全一致（且可通过rviz查看相机的RGB/Depth图像流）
2. 查看./Examples/sg_slam_ros_rgbd以下代码
```c++
Camera_Pose = mpSLAM->TrackRGBD(cv_ptrRGB->image,cv_ptrD->image,cv_ptrRGB->header.stamp.toSec());
```
中cv_ptrRGB->image,cv_ptrD->image等图像变量确实收到了正确的图像流（可以通过cv::imshow等方式显示一下这些图像），然后再排查其他。
3. 可尝试，g2o和sg-slam编译时添加或删除-march=native（从报错信息来看是ROS队列出问题，这里应该影响不大，但可尝试）
```cmake
# Compiler specific options for gcc
SET(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -O3 -march=native")#
SET(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -O3 -march=native")#
```

qqqaah, 2023-09-14:
我换了个系统运行程序，是可行的，或许我的ros系统有点问题，我会尝试更换ros系统，非常感谢您的帮助！

closed 2023-09-14 by qqqaah