silencht/SG-SLAM #10: catkin_make --pkg sg-slam 编译失败，是因为pcl版本的问题吗，请问作者使用的pcl版本

lz-e, 2023-09-20:
编译信息：
[ 20%] Building CXX object sg-slam/CMakeFiles/sg-slam.dir/src/ObjectDatabase.cc.o
[ 23%] Building CXX object sg-slam/CMakeFiles/sg-slam.dir/src/PointcloudMapping.cc.o
In file included from /home/lz/catkin_ws/src/SG-SLAM/src/sg-slam/src/PointcloudMapping.cc:6:0:
/usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:202:14: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
         std::auto_ptr<CloudViewer_impl> impl_;
              ^~~~~~~~
In file included from /usr/include/c++/7/memory:80:0,
                 from /usr/include/c++/7/thread:39,
                 from /home/lz/catkin_ws/src/SG-SLAM/src/sg-slam/include/System.h:26,
                 from /home/lz/catkin_ws/src/SG-SLAM/src/sg-slam/include/PointcloudMapping.h:55,
                 from /home/lz/catkin_ws/src/SG-SLAM/src/sg-slam/src/PointcloudMapping.cc:1:
/usr/include/c++/7/bits/unique_ptr.h:51:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
CMakeFiles/Makefile2:4254: recipe for target 'sg-slam/CMakeFiles/sg-slam.dir/all' failed
make[1]: *** [sg-slam/CMakeFiles/sg-slam.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

silencht, 2023-09-20:
my libpcl-dev version is 1.8.1, eigen version is 3.3.4
It looks like the PCL compilation just came up with a **warning** rather than an **error**. Is there an **error** in some other location?

lz-e, 2023-09-20:
> It looks like the PCL compilation just came up with a **warning** rather than an **error**. Is there an **error** in some other location?
lz@lz-Lenovo-Legion-Y7000:~/catkin_ws/src/SG-SLAM$ catkin_make --pkg sg-slam
Base path: /home/lz/catkin_ws/src/SG-SLAM
Source space: /home/lz/catkin_ws/src/SG-SLAM/src
Build space: /home/lz/catkin_ws/src/SG-SLAM/build
Devel space: /home/lz/catkin_ws/src/SG-SLAM/devel
Install space: /home/lz/catkin_ws/src/SG-SLAM/install
####
#### Running command: "make cmake_check_build_system" in "/home/lz/catkin_ws/src/SG-SLAM/build"
####
####
#### Running command: "make -j12 -l12" in "/home/lz/catkin_ws/src/SG-SLAM/build/sg-slam"
####
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  8%] Built target cv_bridge
[ 17%] Built target image_geometry
Consolidate compiler generated dependencies of target sg-slam
make[2]: *** No rule to make target '/home/lz/catkin_ws/src/SG-SLAM/src/sg-slam/Thirdparty/g2o/lib/libg2o.so', needed by '/home/lz/catkin_ws/src/SG-SLAM/src/sg-slam/lib/libsg-slam.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 20%] Building CXX object sg-slam/CMakeFiles/sg-slam.dir/src/ObjectDatabase.cc.o
[ 23%] Building CXX object sg-slam/CMakeFiles/sg-slam.dir/src/PointcloudMapping.cc.o
In file included from /home/lz/catkin_ws/src/SG-SLAM/src/sg-slam/src/PointcloudMapping.cc:6:0:
/usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:202:14: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
         std::auto_ptr<CloudViewer_impl> impl_;
              ^~~~~~~~
In file included from /usr/include/c++/7/memory:80:0,
                 from /usr/include/c++/7/thread:39,
                 from /home/lz/catkin_ws/src/SG-SLAM/src/sg-slam/include/System.h:26,
                 from /home/lz/catkin_ws/src/SG-SLAM/src/sg-slam/include/PointcloudMapping.h:55,
                 from /home/lz/catkin_ws/src/SG-SLAM/src/sg-slam/src/PointcloudMapping.cc:1:
/usr/include/c++/7/bits/unique_ptr.h:51:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
CMakeFiles/Makefile2:4254: recipe for target 'sg-slam/CMakeFiles/sg-slam.dir/all' failed
make[1]: *** [sg-slam/CMakeFiles/sg-slam.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2

这是所有的编译信息了，没有其他地方报错，我的是pcl 1.7 eigen3.3.1。

silencht, 2023-09-20:
> Consolidate compiler generated dependencies of target sg-slam
> make[2]: *** No rule to make target '/home/lz/catkin_ws/src/SG-SLAM/src/sg-slam/Thirdparty/g2o/lib/libg2o.so', needed by '/home/lz/catkin_ws/src/SG-SLAM/src/sg-slam/lib/libsg-slam.so'. Stop.
这里可能是关键所在。
你成功运行过./ThirdpartyBuild.sh了吗？该脚本下主要编译DBoW2、g2o、ncnn（还需手动编译配置）等库。

lz-e, 2023-09-20:
> 这里可能是关键所在。 你成功运行过./ThirdpartyBuild.sh了吗？该脚本下主要编译DBoW2、g2o、ncnn（还需手动编译配置）等库。
谢谢！确实因为g2o没有编译成功，因为找不到eigen3，已解决问题。现在编译成功了。

closed 2023-10-31 by silencht