silencht/SG-SLAM #12: catkin_make --pkg sg-slam 编译失败error: static assertion failed: std::map must have the same value_type as its allocator

CloudsRL, 2023-10-09:
catkin_make --pkg sg-slam编译出现错误，在回环检测和后端优化.cc中，出现地图类型不匹配的问题QAQ
其他环境都是按照要求配置的，但是opencv本地用的3.2.0版本，gcc9.4.0，g++9.4.0版本。求助
——————————————————————————分界线——————————————————————————
In file included from /usr/include/c++/9/map:61,
                 from /home/wang/code/SG-SLAM/src/sg-slam/Thirdparty/DBoW2/DBoW2/BowVector.h:14,
                 from /home/wang/code/SG-SLAM/src/sg-slam/include/Frame.h:27,
                 from /home/wang/code/SG-SLAM/src/sg-slam/include/MapPoint.h:25,
                 from /home/wang/code/SG-SLAM/src/sg-slam/include/KeyFrame.h:24,
                 from /home/wang/code/SG-SLAM/src/sg-slam/include/LoopClosing.h:24,
                 from /home/wang/code/SG-SLAM/src/sg-slam/src/LoopClosing.cc:21:
/usr/include/c++/9/bits/stl_map.h: In instantiation of ‘class std::map<ORB_SLAM2::KeyFrame*, g2o::Sim3, std::less<ORB_SLAM2::KeyFrame*>, Eigen::aligned_allocator<std::pair<const ORB_SLAM2::KeyFrame*, g2o::Sim3> > >’:
/home/wang/code/SG-SLAM/src/sg-slam/src/LoopClosing.cc:438:21:   required from here
/usr/include/c++/9/bits/stl_map.h:122:71: error: static assertion failed: std::map must have the same value_type as its allocator
  122 |       static_assert(is_same<typename _Alloc::value_type, value_type>::value,
      |                                                                       ^~~~~
——————————————————————————分界线——————————————————————————
In file included from /usr/include/c++/9/map:61,
                 from /home/wang/code/SG-SLAM/src/sg-slam/Thirdparty/DBoW2/DBoW2/BowVector.h:14,
                 from /home/wang/code/SG-SLAM/src/sg-slam/include/KeyFrame.h:25,
                 from /home/wang/code/SG-SLAM/src/sg-slam/include/MapPoint.h:24,
                 from /home/wang/code/SG-SLAM/src/sg-slam/include/Map.h:24,
                 from /home/wang/code/SG-SLAM/src/sg-slam/include/Optimizer.h:24,
                 from /home/wang/code/SG-SLAM/src/sg-slam/src/Optimizer.cc:21:
/usr/include/c++/9/bits/stl_map.h: In instantiation of ‘class std::map<ORB_SLAM2::KeyFrame*, g2o::Sim3, std::less<ORB_SLAM2::KeyFrame*>, Eigen::aligned_allocator<std::pair<const ORB_SLAM2::KeyFrame*, g2o::Sim3> > >’:
/home/wang/code/SG-SLAM/src/sg-slam/src/Optimizer.cc:818:37:   required from here
/usr/include/c++/9/bits/stl_map.h:122:71: error: static assertion failed: std::map must have the same value_type as its allocator
  122 |       static_assert(is_same<typename _Alloc::value_type, value_type>::value,
      |                                                                       ^~~~~

silencht, 2023-10-09:
参考一下https://zhuanlan.zhihu.com/p/218019316

CloudsRL, 2023-10-10:
非常感谢！
我是用评论区的解决方式解决的，也就是把LoopClosing.cc错误的对应代码改成下面这样：
Eigen::aligned_allocator<std::pair<KeyFrame *const, g2o::Sim3> > > KeyFrameAndPose;
但我想请问一下，这种可能是由什么原因引起的呢？我在编译DynaSlam的时候也遇到了类似的错误

closed 2023-10-17 by CloudsRL