silencht/SG-SLAM #14: 为何rviz中无法显示全局的八叉树地图？

jiqiang2022, 2023-10-30:
你好，根据你的教程，我已经将代码运行成功，在运行run_tum_walking_xyz数据集时，rviz中只显示了红蓝方块，而不显示全局的八叉树地图，好像局部八叉树地图可以显示，是否可以解答，谢谢！

silencht, 2023-10-30:
所有可以定制的参数在README中已经说明。
能否提供系统启动的所有信息？（我猜测可能是octomap_server功能包没有启动，或者出现问题？）

jiqiang2022, 2023-10-30:
> 所有可以定制的参数在README中已经说明。 能否提供系统启动的所有信息？（我猜测可能是octomap_server功能包没有启动，或者出现问题？）

确实，我之前没有注意，运行octomap_server功能包roslaunch octomap.launch
出现：
[ERROR] [1698668587.135397414]: Colored map requested in launch file - node not running/compiled to support colors, please define COLOR_OCTOMAP_SERVER and recompile or launch the octomap_color_server node
[ WARN] [1698668587.157050996]: Nothing to publish, octree is empty
这种情况是怎么解决

silencht, 2023-10-30:
由于工程下的octomap_server的COLOR_OCTOMAP_SERVER宏定义已经打开。
所以我猜测可能是你launch的octomap_server是系统之前安装的，而不是本工程下的octomap_server。
可以尝试一下在进行
```bash
cd your_sg-slam_path/src/octomap_server/launch
roslaunch octomap.launch
```
命令前，运行一次，将当前功能包目录添加到系统变量。
```bash
source devel/setup.bash
```
如果还是不行，尝试卸载之前功能包（此步骤需谨慎执行）？
```bash
sudo apt-get install ros-melodic-octomap-server
```

jiqiang2022, 2023-10-30:
> 由于工程下的octomap_server的COLOR_OCTOMAP_SERVER宏定义已经打开。 所以我猜测可能是你launch的octomap_server是系统之前安装的，而不是本工程下的octomap_server。 可以尝试以下步骤
> 
> ```shell
> #命令前，运行一次，将当前功能包目录添加到系统变量。
> source devel/setup.bash
> cd your_sg-slam_path/src/octomap_server/launch
> roslaunch octomap.launch
> ```
> 
> 如果还是不行，尝试卸载之前功能包（此步骤需谨慎执行）？
> 
> ```shell
> sudo apt-get remove ros-melodic-octomap-server
> ```
根据你说的进行操作，问题已经解决，十分感谢！

closed 2023-10-30 by silencht