silencht/SG-SLAM #19: catkin_make --pkg sg-slam 编译出错

A-ASTON, 2023-12-13:
错误如下，请问如何解决？感谢
[ 94%] Linking CXX executable /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/Examples/sg_slam_tum
[ 94%] Linking CXX executable /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/Examples/sg_slam_ros_rgbd
/usr/bin/ld: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/lib/libsg-slam.so: undefined reference to `ORB_SLAM2::Sim3Solver::GetEstimatedTranslation()'
/usr/bin/ld: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/lib/libsg-slam.so: undefined reference to `ORB_SLAM2::Sim3Solver::SetRansacParameters(double, int, int)'
/usr/bin/ld: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/lib/libsg-slam.so: undefined reference to `ORB_SLAM2::Sim3Solver::GetEstimatedRotation()'
/usr/bin/ld: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/lib/libsg-slam.so: undefined reference to `ORB_SLAM2::Sim3Solver::GetEstimatedScale()'
/usr/bin/ld: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/lib/libsg-slam.so: undefined reference to `ORB_SLAM2::Sim3Solver::iterate(int, bool&, std::vector<bool, std::allocator<bool> >&, int&)'
/usr/bin/ld: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/lib/libsg-slam.so: undefined reference to `ORB_SLAM2::Sim3Solver::Sim3Solver(ORB_SLAM2::KeyFrame*, ORB_SLAM2::KeyFrame*, std::vector<ORB_SLAM2::MapPoint*, std::allocator<ORB_SLAM2::MapPoint*> > const&, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [sg-slam/CMakeFiles/sg_slam_tum.dir/build.make:267: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/Examples/sg_slam_tum] Error 1
make[1]: *** [CMakeFiles/Makefile2:4415: sg-slam/CMakeFiles/sg_slam_tum.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/lib/libsg-slam.so: undefined reference to `ORB_SLAM2::Sim3Solver::GetEstimatedTranslation()'
/usr/bin/ld: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/lib/libsg-slam.so: undefined reference to `ORB_SLAM2::Sim3Solver::SetRansacParameters(double, int, int)'
/usr/bin/ld: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/lib/libsg-slam.so: undefined reference to `ORB_SLAM2::Sim3Solver::GetEstimatedRotation()'
/usr/bin/ld: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/lib/libsg-slam.so: undefined reference to `ORB_SLAM2::Sim3Solver::GetEstimatedScale()'
/usr/bin/ld: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/lib/libsg-slam.so: undefined reference to `ORB_SLAM2::Sim3Solver::iterate(int, bool&, std::vector<bool, std::allocator<bool> >&, int&)'
/usr/bin/ld: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/lib/libsg-slam.so: undefined reference to `ORB_SLAM2::Sim3Solver::Sim3Solver(ORB_SLAM2::KeyFrame*, ORB_SLAM2::KeyFrame*, std::vector<ORB_SLAM2::MapPoint*, std::allocator<ORB_SLAM2::MapPoint*> > const&, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [sg-slam/CMakeFiles/sg_slam_ros_rgbd.dir/build.make:267: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/Examples/sg_slam_ros_rgbd] Error 1
make[1]: *** [CMakeFiles/Makefile2:4386: sg-slam/CMakeFiles/sg_slam_ros_rgbd.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j12 -l12" failed

A-ASTON, 2023-12-13:
单独拎一个出来看就是ld无法找到so库中的函数？
/usr/bin/ld: /home/aston/projects/dpSLAM/SG-SLAM/src/sg-slam/lib/libsg-slam.so: undefined reference to ORB_SLAM2::Sim3Solver::GetEstimatedTranslation()'

silencht, 2023-12-13:
使用ldd命令查看一下libsg-slam.so依赖的库是否有not found?

A-ASTON, 2023-12-14:
没有发现not found的库

silencht, 2023-12-15:
无法理解出现该错误的原因。
或许你可以在CMakeLists.txt中相应位置尝试查看并添加 your_project_path/sg-slam/include 路径（Sim3Solver.h文件所在目录），以帮助ld链接器寻找头文件函数声明。

closed 2024-04-09 by silencht