silencht/SG-SLAM #20: 运行./run_tum_walking_xyz.sh无结果

QQQ300kuai, 2023-12-15:
你好，运行问题命令为（数据集存放位置改变了）
`./Examples/sg_slam_tum Vocabulary/ORBvoc.bin Examples/TUM3.yaml /home/slam/datasets/rgbd_dataset_freiburg3_walking_xyz_validation /home/slam/datasets/rgbd_dataset_freiburg3_walking_xyz_validation/associations.txt
`

[image]

当我直接输入命令运行，也一样无反应

[image]

QQQ300kuai, 2023-12-15:
这是我的topic list。有没有什么解决办法，感谢
[image]

silencht, 2023-12-15:
如果仅输入 `./Examples/sg_slam_tum`命令会输出Usage: ./rgbd_tum path_to_vocabulary path_to_settings path_to_sequence path_to_association，那么请继续仔细检查后续的参数项

QQQ300kuai, 2023-12-15:
我重新检查了参数，应该是没问题的。
现在运行显示的是段错误。
[image]

QQQ300kuai, 2023-12-15:
我ORB-SLAM3是能正常运行的

silencht, 2023-12-15:
请使用
```
gdb ./Examples/sg_slam_tum
```
进入调试模式，
然后输入
```
run + 剩余参数（path_to_vocabulary path_to_settings path_to_sequence path_to_association）
```
启动slam。
查看段错误发生位置。产生段错误后，可使用`bt`命令查看回溯函数调用堆栈。
如果堆栈回溯定位不清晰，可在CMakeLists.txt中设置Debug模式，编译程序后，再次进行以上步骤的调试。

QQQ300kuai, 2023-12-18:
这是显示的结果，可以从哪块入手，解决段错误呢。不太熟悉
[image]

silencht, 2023-12-18:
看样子可能是cv::file读取yaml文件读取路径有问题，文件打开失败。

QQQ300kuai, 2023-12-18:
我查看了，传入yaml文件位置是准确的

QQQ300kuai, 2023-12-18:
CMakeLists.txt中设置Debug模式后，这是更详细的信息
[image]

QQQ300kuai, 2023-12-18:
将所有参数文件路径均替换为全局路径依旧错误，错误信息一样。文件内容也是完好的

在代码62行中，发生段错误。
[image]

我重新测试了62行这段打开文件代码，是能正确运行的
[image]

silencht, 2023-12-18:
奇怪的错误。

在System.cc第62行（发生段错误）之前使用std::cout打印strSettingsFile.c_str()值，查看值是否正常。并与你的测试代码进行对比。

QQQ300kuai, 2023-12-18:
我发现问题所在了，我电脑中是有opencv 4.2和3.3.1,修改了sg-slam的CMakeLists.txt为3.3.1，而cv_bridge使用的还是4.2，版本冲突造成，都改为3.3.1就能运行了。

但我又遇到另一个问题，rviz八叉树地图不能显示

[image]

`rviz -d /home/slam/SG-SLAM/src/sg-slam/Examples/rvizconfig.rviz  错误信息
`
[image]

QQQ300kuai, 2023-12-18:
[image]

silencht, 2023-12-19:
可能是ROS版本导致的tf/tf2迁移问题吧。
尝试把https://github.com/silencht/SG-SLAM/blob/7f34661d91f20a22995adad5813de7d964e859f3/src/octomap_server/launch/octomap.launch#L11 以及example目录下等与坐标系相关的/map均修改为map后再次编译运行。

QQQ300kuai, 2023-12-19:
有帮助，八叉树地图可以显示，但语义对象度量地图无法显示，是哪个地方的/map没改的问题吗。
[image]

还有稠密点云图这样是正常的吗？运行的是gbd_dataset_freiburg3_walking_xyz序列
[image]

QQQ300kuai, 2023-12-19:
> 可能是ROS版本导致的tf/tf2迁移问题吧。 尝试把
> 
> https://github.com/silencht/SG-SLAM/blob/7f34661d91f20a22995adad5813de7d964e859f3/src/octomap_server/launch/octomap.launch#L11
> 
> 以及example目录下等与坐标系相关的/map均修改为map后再次编译运行。

是只有这两个地方的/map需要修改吗，我只修改了这两个地方，/map修改为了map
https://github.com/silencht/SG-SLAM/blob/7f34661d91f20a22995adad5813de7d964e859f3/src/sg-slam/Examples/rgbd_tum.cc#L207

https://github.com/silencht/SG-SLAM/blob/7f34661d91f20a22995adad5813de7d964e859f3/src/octomap_server/launch/octomap.launch#L13

QQQ300kuai, 2023-12-19:
可以正常显示了。还有SG-SLAM/src/sg-slam/src/PointcloudMapping.cc里的/map没有修改，修改后语义对象度量地图和稠密地图都能显示。是ROS版本导致的tf/tf2迁移问题

在其他issue中看到了更详细的出现/map的位置https://github.com/silencht/SG-SLAM/issues/1#issuecomment-1587559048

想知道SG-SLAM是否可以用在室外场景中，作者有跑过KITTI等室外数据集吗？

silencht, 2023-12-19:
> 还有稠密点云图这样是正常的吗？运行的是gbd_dataset_freiburg3_walking_xyz序列

看起来点云很稀疏，点云比较稀疏的原因可能是（你修改过的）分辨率等参数设置的比较低？正常效果可参考README中的视频链接。

> 想知道SG-SLAM是否可以用在室外场景中，作者有跑过KITTI等室外数据集吗？

这取决于你的RGB-D相机设备的工作环境支持情况，KITTI好像是双目数据集，我没有跑过它。
（因为ORB-SLAM系统对双目和RGB-D输入的处理差异比较小，故系统其实可以适配至双目，但这需要一定的工程量）

closed 2023-12-21 by silencht