silencht/SG-SLAM #21: 运行数据集时，窗口闪退

zpj159, 2023-12-26:
作者您好，很荣幸看到这么优秀的语言地图构建的开源代码，在我按照您给出的操作中，运行run_tum_walking_xyz.sh这个文件时，窗口闪退，通过使用gdb，查询到错误是Thread 1 "sg_slam_tum" received signal SIGSEGV, Segmentation fault.
0x00007fffef36b098 in g2o::EdgeSE3ProjectXYZOnlyPose::linearizeOplus() ()
   from /home/1/SLAM/SG-SLAM/src/sg-slam/Thirdparty/g2o/lib/libg2o.so，这是因为我的g2o的库出了问题吗，还望您给予帮助，非常感谢！

zpj159, 2023-12-26:
gdb ./Examples/sg_slam_tum
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./Examples/sg_slam_tum...(no debugging symbols found)...done.
(gdb) run ./Vocabulary/ORBvoc.bin ./Examples/TUM3.yaml /home/1/SLAM/TUMDataset/rgbd_dataset_freiburg3_walking_xyz /home/1/SLAM/TUMDataset/rgbd_dataset_freiburg3_walking_xyz/associations.txt
Starting program: /home/1/SLAM/SG-SLAM/src/sg-slam/Examples/sg_slam_tum ./Vocabulary/ORBvoc.bin ./Examples/TUM3.yaml /home/1/SLAM/TUMDataset/rgbd_dataset_freiburg3_walking_xyz /home/1/SLAM/TUMDataset/rgbd_dataset_freiburg3_walking_xyz/associations.txt
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fff9f261700 (LWP 31784)]
[New Thread 0x7fff9d805700 (LWP 31788)]
[New Thread 0x7fff9d004700 (LWP 31789)]
[New Thread 0x7fff9c803700 (LWP 31790)]
[New Thread 0x7fff97fff700 (LWP 31791)]

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

is map construction consider dynamic: 1
is global pointcloud reconstruction: 0
is octomap semanticObject map construction: 1
camera_valid_depth_Min is 0.5
camera_valid_depth_Max is 5
Sor_Local_MeanK is 50
Sor_Local_StddevMulThresh is 3
Voxel_Local_LeafSize is 0.01
Sor_Global_MeanK is 50
Sor_Global_StddevMulThresh is 3
Voxel_Global_LeafSize is 0.01
Detector3D_Sor_MeanK is 50
Detector3D_Sor_StddevMulThresh is 1
Detector3D_Voxel_LeafSize is 0.01
Detect3D_EuclideanClusterTolerance is 0.02
Detect3D_EuclideanClusterMinSize is 1000
Detect3D_EuclideanClusterMaxSize is 30000
Detect3D_DetectSimilarCompareRatio is 0.1
global_pc_update_kf_threshold is 25
[New Thread 0x7fff960dd700 (LWP 31794)]
Start PointCloudMapping Viewer...
[New Thread 0x7fff7d6d7700 (LWP 31796)]
[New Thread 0x7fff7ced6700 (LWP 31797)]
[New Thread 0x7fff5dfff700 (LWP 31799)]
[New Thread 0x7fff5d7fe700 (LWP 31800)]
[New Thread 0x7fff5cffd700 (LWP 31801)]
[New Thread 0x7fff51fff700 (LWP 31802)]
[New Thread 0x7fff517fe700 (LWP 31803)]

Camera Parameters: 
- fx: 535.4
- fy: 539.2
- cx: 320.1
- cy: 247.6
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 2.98842
[New Thread 0x7fff506c3700 (LWP 31809)]
[New Thread 0x7fff4fec2700 (LWP 31810)]

detection_confidence_threshold is 0.9

dynamic_detection_confidence_threshold is 0.01
[New Thread 0x7fff4f6c1700 (LWP 31811)]
[New Thread 0x7fff4eec0700 (LWP 31812)]
[New Thread 0x7fff4e6bf700 (LWP 31813)]
[New Thread 0x7fff4debe700 (LWP 31814)]
Detector2D Thread start ...
[New Thread 0x7fff4d6bd700 (LWP 31815)]

-------
Start processing sequence ...
Images in the sequence: 827

[New Thread 0x7fff4cebc700 (LWP 31816)]
[New Thread 0x7fff4638b700 (LWP 31817)]
[New Thread 0x7fff45389700 (LWP 31820)]
[New Thread 0x7fff45b8a700 (LWP 31819)]
[New Thread 0x7fff37fff700 (LWP 31822)]
[New Thread 0x7fff44b88700 (LWP 31821)]
[New Thread 0x7fff377fe700 (LWP 31823)]
[New Thread 0x7fff36ffd700 (LWP 31824)]
[New Thread 0x7fff367fc700 (LWP 31825)]
[New Thread 0x7fff35ffb700 (LWP 31826)]
[New Thread 0x7fff357fa700 (LWP 31827)]
New map created with 717 points

Thread 1 "sg_slam_tum" received signal SIGSEGV, Segmentation fault.
0x00007fffef36b098 in g2o::EdgeSE3ProjectXYZOnlyPose::linearizeOplus() ()
   from /home/1/SLAM/SG-SLAM/src/sg-slam/Thirdparty/g2o/lib/libg2o.so

silencht, 2023-12-26:
可能得解决办法，请尝试：
在主工程目录和DBoW、g2o等第三方库的多个CMakeLists.txt中，在`-O3`后统一添加（或删除）`-march=native`标志，并清除build文件夹下的所有cmake缓存文件，重新完整编译工程+运行。

```cmake
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  -O3 -march=native ")
```

from https://github.com/raulmur/ORB_SLAM2/pull/844

zpj159, 2023-12-26:
是的，按照作者您给出的方法，解决了上述问题，非常感谢

closed 2023-12-26 by zpj159