silencht/SG-SLAM #22: Can't get pointcloud map or semantic map via AstraPro camera

Cooo1der, 2023-12-26:
Thank your for your excellent work! Howerver, when I ran the project  code with my AstraPro camera, I was unable to create point cloud map or semantic map.
Coud you please give some advice to solve these problems? It bothers me for a long time... 

Here are some of my main outputs:
**1. roslaunch astra_camera astra_pro.launch**
SUMMARY
========
PARAMETERS
 * /camera/camera/camera_name: camera
 * /camera/camera/color_depth_synchronization: True
 * /camera/camera/color_format: RGB
 * /camera/camera/color_fps: 30
 * /camera/camera/color_height: 480
 * /camera/camera/color_info_uri: file:///home/slam...
 * /camera/camera/color_roi_height: -1
 * /camera/camera/color_roi_width: -1
 * /camera/camera/color_roi_x: -1
 * /camera/camera/color_roi_y: -1
 * /camera/camera/color_width: 640
 * /camera/camera/connection_delay: 100
 * /camera/camera/depth_align: True
 * /camera/camera/depth_format: Y11
 * /camera/camera/depth_fps: 30
 * /camera/camera/depth_height: 480
 * /camera/camera/depth_roi_height: -1
 * /camera/camera/depth_roi_width: -1
 * /camera/camera/depth_roi_x: -1
 * /camera/camera/depth_roi_y: -1
 * /camera/camera/depth_scale: 1
 * /camera/camera/depth_width: 640
 * /camera/camera/device_num: 1
 * /camera/camera/enable_color: True
 * /camera/camera/enable_d2c_viewer: True
 * /camera/camera/enable_depth: True
 * /camera/camera/enable_ir: True
 * /camera/camera/enable_point_cloud: True
 * /camera/camera/enable_point_cloud_xyzrgb: True
 * /camera/camera/enable_publish_extrinsic: True
 * /camera/camera/flip_color: True
 * /camera/camera/flip_depth: True
 * /camera/camera/flip_ir: True
 * /camera/camera/ir_format: Y10
 * /camera/camera/ir_fps: 30
 * /camera/camera/ir_height: 480
 * /camera/camera/ir_info_uri: file:///home/slam...
 * /camera/camera/ir_width: 640
 * /camera/camera/oni_log_level: verbose
 * /camera/camera/oni_log_to_console: False
 * /camera/camera/oni_log_to_file: False
 * /camera/camera/product_id: 0
 * /camera/camera/publish_tf: True
 * /camera/camera/serial_number: 
 * /camera/camera/tf_publish_rate: 10.0
 * /camera/camera/use_uvc_camera: True
 * /camera/camera/uvc_camera_format: mjpeg
 * /camera/camera/uvc_flip: False
 * /camera/camera/uvc_product_id: 0x0501
 * /camera/camera/uvc_retry_count: 100
 * /camera/camera/uvc_vendor_id: 0x2bc5
 * /camera/camera/vendor_id: 0
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /camera/
    camera (astra_camera/astra_camera_node)

**2. ./run_astra_pro_camera.sh
is map construction consider dynamic: 1
is global pointcloud reconstruction: 1
is octomap semanticObject map construction: 0
camera_valid_depth_Min is 0.5
camera_valid_depth_Max is 5
Sor_Local_MeanK is 50
Sor_Local_StddevMulThresh is 2
Voxel_Local_LeafSize is 0.01
Sor_Global_MeanK is 50
Sor_Global_StddevMulThresh is 3
Voxel_Global_LeafSize is 0.01
Detector3D_Sor_MeanK is 50
Detector3D_Sor_StddevMulThresh is 1
Detector3D_Voxel_LeafSize is 0.01
Detect3D_EuclideanClusterTolerance is 0.02
Detect3D_EuclideanClusterMinSize is 1000
Detect3D_EuclideanClusterMaxSize is 30000
Detect3D_DetectSimilarCompareRatio is 0.1
global_pc_update_kf_threshold is 25
Start PointCloudMapping Viewer...

Camera Parameters: 
- fx: 601.71
- fy: 612.069
- cx: 327.385
- cy: 182.954
- k1: 0.161383
- k2: -0.277458
- p1: -0.049239
- p2: 0.002004
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 2.65909

detection_confidence_threshold is 0.99

dynamic_detection_confidence_threshold is 0.05
Detector2D Thread start ...
New map created with 794 point
[image]

silencht, 2023-12-27:
If the dataset works well, please check:
- camera Intrinsics parameters (e.g. cx,cy,fx,fy), Deptmap values factor `DepthMapFactor`, etc.   So, why isn't your `cy` around 240.0f ?
- [Subscribe to **appropriate** depth image topics:](https://github.com/silencht/SG-SLAM/blob/7f34661d91f20a22995adad5813de7d964e859f3/src/sg-slam/Examples/ros_rgbd.cc#L74) I was running OK with the TUM dataset, but once I ran actual (Orbbec) hardware RGBD cameras on ROS would fail. I eventually found that the problem was localised to the ROS subscription camera node, the depth image I subscribed to had been `/camera/depth_registered/image_raw`, and the problem solved after I switched to `/camera/depth/image_raw`. 
This is the difference between the two nodes: https://answers.ros.org/question/264378/difference-between-depth-and-depth_registered/
- Use commands such as `std::cout` and `cv::imshow` more often to ensure the correct flow of data

Cooo1der, 2023-12-27:
Thank you for your reply. I tried to recalibrate the internal parameters of the camera, but I found that each time my calibration results were different, even sugnificantly so. Do you know what might be the reason?

silencht, 2023-12-28:
you can find more informations at these website:
- https://docs.opencv.org/4.x/dc/dbb/tutorial_py_calibration.html (this is the official calibration tutorial from OpenCV)
- http://wiki.ros.org/camera_calibration (i use this tool calibrate my camera.)
- https://github.com/orbbec/ros_astra_camera (this is orbbec/astra camera's driver on ros)
- https://www.orbbec.com/developers/

Cooo1der, 2024-01-04:
> I think the correct `cx`, `cy` should be around half of the _width_ and _height_ of the image. you can find more informations at these websites:
> 
> * https://docs.opencv.org/4.x/dc/dbb/tutorial_py_calibration.html (this is the official calibration tutorial from OpenCV)
> * http://wiki.ros.org/camera_calibration (i use this tool calibrate my camera.)
> * https://github.com/orbbec/ros_astra_camera (this is orbbec/astra camera's driver on ros)
> * https://www.orbbec.com/developers/

May I ask how the topic /camera/depth_registered/img_raw was generated? The topic of my camera is only /camera/depth/img_raw. Will this affect the construction of the image

silencht, 2024-01-04:
> * [Subscribe to **appropriate** depth image topics:](https://github.com/silencht/SG-SLAM/blob/7f34661d91f20a22995adad5813de7d964e859f3/src/sg-slam/Examples/ros_rgbd.cc#L74) I was running OK with the TUM dataset, but once I ran actual (Orbbec) hardware RGBD camera on ROS would fail. I eventually found that the problem was localised to the ROS subscription camera node, the depth image I subscribed to had been `/camera/depth_registered/image_raw`, and the problem solved after I switched to `/camera/depth/image_raw`.

This suggestion is only for certain cameras, such as the LeTMC520 camera (actually a cased version of the astra pro) I use. 
These websites below is when I use LeTMC520 camera reference.
- https://github.com/orbbec/ros_astra_camera
- https://github.com/silencht/SLAM_Note/blob/master/system_environment/LeTMC520_Camera.md

closed 2024-04-09 by silencht