silencht/SG-SLAM #23: 可以替换成自己训练的检测模型吗？

QQQ300kuai, 2023-12-27:
可以的话，大致流程是怎样的呢

silencht, 2023-12-27:
本工程下主要相关代码
https://github.com/silencht/SG-SLAM/blob/main/src/sg-slam/src/Detector2D.cc
ncnn例程
https://github.com/Tencent/ncnn/tree/master/examples

QQQ300kuai, 2023-12-27:
您好，在您项目中的mobilenetv3_ssdlite_voc.param和mobilenetv3_ssdlite_voc.bin是自己生成的还是来自于官方仓库？
想要替换自己训练的目标检测模型，除了生成这两个文件，还需要修改其他部分吗，非常感谢

silencht, 2023-12-28:
> 在您项目中的mobilenetv3_ssdlite_voc.param和mobilenetv3_ssdlite_voc.bin是自己生成的还是来自于官方仓库？

当前工程下来自于ncnn官方仓库。

> 想要替换自己训练的目标检测模型，除了生成这两个文件，还需要修改其他部分吗？

根据ncnn/examples例程对应修改本工程下 https://github.com/silencht/SG-SLAM/blob/main/src/sg-slam/src/Detector2D.cc 文件，一方面是加载模型、参数进行推理获得学习结果，另一方面是将学习结果传递到sg-slam。[本质是ncnn的应用问题]

QQQ300kuai, 2023-12-28:
我换成yolo模型后,语义度量地图出现了问题,标签冗余,大小也有问题,有可能是哪方面的问题.我根据ncnn/examples例程只修改了Detector2D.cc文件.
[image]

八叉树地图过滤不干净,这可能是什么原因?是
`detection_confidence_threshold, dynamic_detection_confidence_threshold`
阈值设置的原因吗
[image]

非常感谢~

silencht, 2024-01-02:
是的，如果Detector2D.cc文件修改没问题的话，问题可能出在`detection_confidence_threshold, dynamic_detection_confidence_threshold`等参数的设置。（不过我更倾向于Detector2D.cc传递的信息可能存在问题）

QQQ300kuai, 2024-01-11:
好的，我先试试

closed 2024-01-11 by QQQ300kuai

sm5278, 2024-04-30:
> 好的，我先试试

您好，請問可以分享您的yolo模型吗？
非常感谢

QQQ300kuai, 2024-05-03:
> > 好的，我先试试
> 
> 您好，請問可以分享您的yolo Detector2D.cc是怎样修改吗？ 非常感谢

根据ncnn/examples例程对应修改本工程下 https://github.com/silencht/SG-SLAM/blob/main/src/sg-slam/src/Detector2D.cc 文件，一方面是加载模型、参数进行推理获得学习结果，另一方面是将学习结果传递到sg-slam。按照这个修改就可以了