silencht/SG-SLAM #24: 运行最后一步./run_tum_walking_xyz.sh时异常退出，Segmentation fault

Hstwhale, 2024-01-11:
非常感谢您的优秀SLAM开源工作，在尝试复现这个项目时遇到了一些问题，希望得到您的帮助。

运行最后一步./run_tum_walking_xyz.sh时异常退出，输出Segmentation fault。
用gdb调试输出信息如下：
`
Thread 16 "sg_slam_tum" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff9bff7700 (LWP 6275)]
0x00007ffff73f8017 in ncnn::resize_bilinear_c3(unsigned char const*, int, int, int, unsigned char*, int, int, int) () from /home/ubuntu/SG-SLAM/src/sg-slam/lib/libsg-slam.so
`
看到之前有个issue和我是相同的情况，但gdb输出不是ncnn的问题。我参考您在那个issue的回答，尝试设置-march=native标志后，问题依然没有解决。这个bug请问您有什么解决思路吗？

Hstwhale, 2024-01-11:
`
ubuntu@bad32899ad2f:~/SG-SLAM/src/sg-slam$ gdb ./Examples/sg_slam_tum 
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./Examples/sg_slam_tum...(no debugging symbols found)...done.
(gdb) run Vocabulary/ORBvoc.bin Examples/TUM3.yaml ~/Music/rgbd_dataset_freiburg3_walking_xyz ~/Music/rgbd_dataset_freiburg3_walking_xyz/associations.txt
Starting program: /home/ubuntu/SG-SLAM/src/sg-slam/Examples/sg_slam_tum Vocabulary/ORBvoc.bin Examples/TUM3.yaml ~/Music/rgbd_dataset_freiburg3_walking_xyz ~/Music/rgbd_dataset_freiburg3_walking_xyz/associations.txt
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffc3cac700 (LWP 6260)]
[New Thread 0x7fffc34ab700 (LWP 6261)]
[New Thread 0x7fffc2caa700 (LWP 6262)]
[New Thread 0x7fffc24a9700 (LWP 6263)]

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

is map construction consider dynamic: 1
is global pointcloud reconstruction: 0
is octomap semanticObject map construction: 1
camera_valid_depth_Min is 0.5
camera_valid_depth_Max is 5
Sor_Local_MeanK is 50
Sor_Local_StddevMulThresh is 3
Voxel_Local_LeafSize is 0.01
Sor_Global_MeanK is 50
Sor_Global_StddevMulThresh is 3
Voxel_Global_LeafSize is 0.01
Detector3D_Sor_MeanK is 50
Detector3D_Sor_StddevMulThresh is 1
Detector3D_Voxel_LeafSize is 0.01
Detect3D_EuclideanClusterTolerance is 0.02
Detect3D_EuclideanClusterMinSize is 1000
Detect3D_EuclideanClusterMaxSize is 30000
Detect3D_DetectSimilarCompareRatio is 0.1
global_pc_update_kf_threshold is 25
[New Thread 0x7fffa56d7700 (LWP 6265)]
Start PointCloudMapping Viewer...

Camera Parameters: 
- fx: 535.4
- fy: 539.2
- cx: 320.1
- cy: 247.6
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 2.98842
[New Thread 0x7fffa4ed6700 (LWP 6266)]
[New Thread 0x7fff9ffff700 (LWP 6267)]

detection_confidence_threshold is 0.9

dynamic_detection_confidence_threshold is 0.01
[New Thread 0x7fff9f7fe700 (LWP 6268)]
[New Thread 0x7fff9effd700 (LWP 6269)]
[New Thread 0x7fff9e7fc700 (LWP 6270)]
[New Thread 0x7fff9dffb700 (LWP 6271)]
[New Thread 0x7fff9d7fa700 (LWP 6272)]
[New Thread 0x7fff9cff9700 (LWP 6273)]
[New Thread 0x7fff9c7f8700 (LWP 6274)]
[New Thread 0x7fff9bff7700 (LWP 6275)]
Detector2D Thread start ...
[New Thread 0x7fff9b7f6700 (LWP 6276)]

-------
Start processing sequence ...
Images in the sequence: 833

[New Thread 0x7fff8bdd5700 (LWP 6277)]

Thread 16 "sg_slam_tum" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff9bff7700 (LWP 6275)]
0x00007ffff73f8017 in ncnn::resize_bilinear_c3(unsigned char const*, int, int, int, unsigned char*, int, int, int) () from /home/ubuntu/SG-SLAM/src/sg-slam/lib/libsg-slam.so
`

silencht, 2024-01-11:
> 0x00007ffff73f8017 in ncnn::resize_bilinear_c3(unsigned char const*, int, int, int, unsigned char*, int, int, int) () from /home/ubuntu/SG-SLAM/src/sg-slam/lib/libsg-slam.so
gdb输出定位到ncnn::resize_bilinear_c3，建议搜寻一下ncnn的相关issues

gdb中使用bt命令查看更多堆栈获取信息.

Hstwhale, 2024-01-11:
这是相关的gdb输出，请问有什么解决思路吗？
这是gdb定位后的输出：
`
(gdb) run Vocabulary/ORBvoc.bin Examples/TUM3.yaml ~/Music/rgbd_dataset_freiburg3_walking_xyz ~/Music/rgbd_dataset_freiburg3_walking_xyz/associations.txt
Starting program: /home/ubuntu/SG-SLAM/src/sg-slam/Examples/sg_slam_tum Vocabulary/ORBvoc.bin Examples/TUM3.yaml ~/Music/rgbd_dataset_freiburg3_walking_xyz ~/Music/rgbd_dataset_freiburg3_walking_xyz/associations.txt
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffc3cac700 (LWP 13065)]
[New Thread 0x7fffc34ab700 (LWP 13066)]
[New Thread 0x7fffc2caa700 (LWP 13067)]
[New Thread 0x7fffc24a9700 (LWP 13068)]

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

is map construction consider dynamic: 1
is global pointcloud reconstruction: 0
is octomap semanticObject map construction: 1
camera_valid_depth_Min is 0.5
camera_valid_depth_Max is 5
Sor_Local_MeanK is 50
Sor_Local_StddevMulThresh is 3
Voxel_Local_LeafSize is 0.01
Sor_Global_MeanK is 50
Sor_Global_StddevMulThresh is 3
Voxel_Global_LeafSize is 0.01
Detector3D_Sor_MeanK is 50
Detector3D_Sor_StddevMulThresh is 1
Detector3D_Voxel_LeafSize is 0.01
Detect3D_EuclideanClusterTolerance is 0.02
Detect3D_EuclideanClusterMinSize is 1000
Detect3D_EuclideanClusterMaxSize is 30000
Detect3D_DetectSimilarCompareRatio is 0.1
global_pc_update_kf_threshold is 25
[New Thread 0x7fffa56d7700 (LWP 13069)]
Start PointCloudMapping Viewer...

Camera Parameters: 
- fx: 535.4
- fy: 539.2
- cx: 320.1
- cy: 247.6
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 2.98842
[New Thread 0x7fffa4ed6700 (LWP 13070)]
[New Thread 0x7fff9ffff700 (LWP 13071)]

detection_confidence_threshold is 0.9

dynamic_detection_confidence_threshold is 0.01
[... 56 lines cut ...]
  what():  OpenCV(3.4.15) /home/ubuntu/opencv/modules/video/src/lkpyramid.cpp:734: error: (-215:Assertion failed) img.depth() == CV_8U && winSize.width > 2 && winSize.height > 2 in function 'buildOpticalFlowPyramid'

Thread 1 "sg_slam_tum" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffff7f41f00 (LWP 13063)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff3f09e87 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff3f0b7f1 in __GI_abort () at abort.c:79
#2  0x00007ffff48fe957 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff4904ae6 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff4904b21 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff4904d54 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff6e96eaa in cv::error(cv::Exception const&) () at /usr/local/lib/libopencv_core.so.3.4
#7  0x00007ffff6e984af in cv::error(int, cv::String const&, char const*, char const*, int) () at /usr/local/lib/libopencv_core.so.3.4
#8  0x00007ffff2bc9f53 in cv::errorNoReturn(int, cv::String const&, char const*, char const*, int) [clone .constprop.127] () at /usr/local/lib/libopencv_video.so.3.4
#9  0x00007ffff2bcd5c7 in cv::buildOpticalFlowPyramid(cv::_InputArray const&, cv::_OutputArray const&, cv::Size_<int>, int, bool, int, int, bool) () at /usr/local/lib/libopencv_video.so.3.4
#10 0x00007ffff2bcf5ff in cv::(anonymous namespace)::SparsePyrLKOpticalFlowImpl::calc(cv::_InputArray const&, cv::_InputArray const&, cv::_InputArray const&, cv::_InputOutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&) () at /usr/local/lib/libopencv_video.so.3.4
#11 0x00007ffff2bcd843 in cv::calcOpticalFlowPyrLK(cv::_InputArray const&, cv::_InputArray const&, cv::_InputArray const&, cv::_InputOutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::Size_<int>, int, cv::TermCriteria, int, double) () at /usr/local/lib/libopencv_video.so.3.4
#12 0x00007ffff736a5b6 in ORB_SLAM2::Frame::RmDynamicPointWithSemanticAndGeometry(cv::Mat const&, cv::Mat const&) () at /home/ubuntu/SG-SLAM/src/sg-slam/lib/libsg-slam.so
#13 0x00007ffff736bac0 in ORB_SLAM2::Frame::Frame(ORB_SLAM2::Tracking*, cv::Mat&, cv::Mat const&, double const&, ORB_SLAM2::ORBextractor*, DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>*, cv::Mat&, cv::Mat&, float const&, float const&, ORB_SLAM2::Frame&) () at /home/ubuntu/SG-SLAM/src/sg-slam/lib/libsg-slam.so
#14 0x00007ffff72dbd50 in ORB_SLAM2::Tracking::GrabImageRGBD(cv::Mat const&, cv::Mat const&, double const&) () at /home/ubuntu/SG-SLAM/src/sg-slam/lib/libsg-slam.so
#15 0x00007ffff72c6c55 in ORB_SLAM2::System::TrackRGBD(cv::Mat const&, cv::Mat const&, double const&) () at /home/ubuntu/SG-SLAM/src/sg-slam/lib/libsg-slam.so
#16 0x000055555540e53f in main ()
`

Hstwhale, 2024-01-11:
> > 0x00007ffff73f8017 in ncnn::resize_bilinear_c3(unsigned char const*, int, int, int, unsigned char*, int, int, int) () from /home/ubuntu/SG-SLAM/src/sg-slam/lib/libsg-slam.so
> 
> gdb输出定位到ncnn::resize_bilinear_c3，建议搜寻一下ncnn的相关issues
> 
> gdb中使用bt命令查看更多堆栈获取信息.

我尝试换了个数据集，没有上述问题了，但是又出现了新的问题：
`(gdb) run Vocabulary/ORBvoc.bin Examples/TUM1.yaml ~/Music/rgbd_dataset_freiburg1_xyz ~/Music/rgbd_dataset_freiburg1_xyz/associations.txt
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/ubuntu/SG-SLAM/src/sg-slam/Examples/sg_slam_tum Vocabulary/ORBvoc.bin Examples/TUM1.yaml ~/Music/rgbd_dataset_freiburg1_xyz ~/Music/rgbd_dataset_freiburg1_xyz/associations.txt
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffc3cac700 (LWP 13786)]
[New Thread 0x7fffc34ab700 (LWP 13787)]
[New Thread 0x7fffc2caa700 (LWP 13788)]
[New Thread 0x7fffc24a9700 (LWP 13789)]

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

is map construction consider dynamic: 1
is global pointcloud reconstruction: 1
is octomap semanticObject map construction: 0
camera_valid_depth_Min is 0.5
camera_valid_depth_Max is 5
Sor_Local_MeanK is 50
Sor_Local_StddevMulThresh is 2
Voxel_Local_LeafSize is 0.01
Sor_Global_MeanK is 50
Sor_Global_StddevMulThresh is 2
Voxel_Global_LeafSize is 0.01
Detector3D_Sor_MeanK is 50
Detector3D_Sor_StddevMulThresh is 1
Detector3D_Voxel_LeafSize is 0.01
Detect3D_EuclideanClusterTolerance is 0.02
Detect3D_EuclideanClusterMinSize is 1000
Detect3D_EuclideanClusterMaxSize is 30000
Detect3D_DetectSimilarCompareRatio is 0.5
global_pc_update_kf_threshold is 25
[New Thread 0x7fffa56d7700 (LWP 13790)]
Start PointCloudMapping Viewer...

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 3.09294
[New Thread 0x7fffa4ed6700 (LWP 13791)]
[New Thread 0x7fff9ffff700 (LWP 13792)]

detection_confidence_threshold is 0.95

dynamic_detection_confidence_threshold is 0.05
[New Thread 0x7fff9f7fe700 (LWP 13793)]
[New Thread 0x7fff9effd700 (LWP 13794)]
[New Thread 0x7fff9e7fc700 (LWP 13795)]
[New Thread 0x7fff9dffb700 (LWP 13796)]
[New Thread 0x7fff9d7fa700 (LWP 13797)]
[New Thread 0x7fff9cff9700 (LWP 13798)]
[New Thread 0x7fff9c7f8700 (LWP 13799)]
[New Thread 0x7fff9bff7700 (LWP 13800)]
Detector2D Thread start ...
[New Thread 0x7fff9b7f6700 (LWP 13801)]

-------
Start processing sequence ...
Images in the sequence: 792

[New Thread 0x7fff8bfff700 (LWP 13802)]
[New Thread 0x7fff8b7fe700 (LWP 13803)]
[New Thread 0x7fff8affd700 (LWP 13804)]
[New Thread 0x7fff8a7fc700 (LWP 13805)]
[New Thread 0x7fff89ffb700 (LWP 13806)]
[New Thread 0x7fff88ff9700 (LWP 13807)]
[New Thread 0x7fff897fa700 (LWP 13808)]
[New Thread 0x7fff73fff700 (LWP 13809)]
[New Thread 0x7fff7bfff700 (LWP 13810)]
[New Thread 0x7fff7b7fe700 (LWP 13811)]
[New Thread 0x7fff7affd700 (LWP 13812)]
[New Thread 0x7fff79ffb700 (LWP 13814)]
[New Thread 0x7fff7a7fc700 (LWP 13813)]
[New Thread 0x7fff78ff9700 (LWP 13816)]
[New Thread 0x7fff797fa700 (LWP 13815)]
[New Thread 0x7fff737fe700 (LWP 13817)]
[New Thread 0x7fff72ffd700 (LWP 13818)]
[New Thread 0x7fff727fc700 (LWP 13819)]
[New Thread 0x7fff71ffb700 (LWP 13820)]
[New Thread 0x7fff717fa700 (LWP 13821)]
[New Thread 0x7fff70ff9700 (LWP 13822)]
[New Thread 0x7fff4bfff700 (LWP 13823)]
[New Thread 0x7fff4b7fe700 (LWP 13824)]
[New Thread 0x7fff4affd700 (LWP 13825)]
[New Thread 0x7fff4a7fc700 (LWP 13826)]
[New Thread 0x7fff49ffb700 (LWP 13827)]
New map created with 833 points
[New Thread 0x7fff497fa700 (LWP 13828)]
[New Thread 0x7fff48ff9700 (LWP 13829)]
[New Thread 0x7fff27fff700 (LWP 13830)]
[New Thread 0x7fff277fe700 (LWP 13831)]
[New Thread 0x7fff26ffd700 (LWP 13832)]
[New Thread 0x7fff267fc700 (LWP 13833)]
[New Thread 0x7fff25ffb700 (LWP 13834)]
[New Thread 0x7fff257fa700 (LWP 13835)]
[New Thread 0x7fff24ff9700 (LWP 13836)]
[New Thread 0x7ffee7fff700 (LWP 13837)]
[New Thread 0x7ffee77fe700 (LWP 13838)]
[New Thread 0x7ffee6ffd700 (LWP 13839)]
[New Thread 0x7ffee67fc700 (LWP 13840)]
[New Thread 0x7ffee5ffb700 (LWP 13841)]
[New Thread 0x7ffee57fa700 (LWP 13842)]
[New Thread 0x7ffee4ff9700 (LWP 13843)]
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.
Thread 6 "sg_slam_tum" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffa56d7700 (LWP 13790)]
__GI___libc_free (mem=0xe9e15) at malloc.c:3113
3113	malloc.c: No such file or directory.
`

closed 2024-01-11 by Hstwhale