silencht/SG-SLAM #25: roslaunch octomap.launch报错不显示颜色

hxy-x, 2024-01-18:
~/SG-SLAM/src/octomap_server/launch$ roslaunch octomap.launch
... logging to /home/hxy/.ros/log/063e30d4-b5ac-11ee-a32d-eddb2c37d7a5/roslaunch-hxy-iCafe-B365M-E-PRO-234278.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hxy-iCafe-B365M-E-PRO:43283/

SUMMARY
========

PARAMETERS
 * /octomap_server/colored_map: True
 * /octomap_server/filter_ground: False
 * /octomap_server/filter_speckles: False
 * /octomap_server/frame_id: map
 * /octomap_server/ground_filter/angle: 0.15
 * /octomap_server/ground_filter/distance: 0.05
 * /octomap_server/ground_filter/plane_distance: 0.05
 * /octomap_server/height_map: False
 * /octomap_server/latch: False
 * /octomap_server/occupancy_max_z: 1.5
 * /octomap_server/occupancy_min_z: -1.5
 * /octomap_server/pointcloud_max_z: 5.0
 * /octomap_server/pointcloud_min_z: -5.0
 * /octomap_server/resolution: 0.03
 * /octomap_server/sensor_model/hit: 0.7
 * /octomap_server/sensor_model/max: 0.97
 * /octomap_server/sensor_model/max_range: -1.0
 * /octomap_server/sensor_model/min: 0.12
 * /octomap_server/sensor_model/miss: 0.4
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    octomap_server (octomap_server/octomap_server_node)

ROS_MASTER_URI=http://localhost:11311

process[octomap_server-1]: started with pid [234292]
[ERROR] [1705566784.095698566]: Colored map requested in launch file - node not running/compiled to support colors, please define COLOR_OCTOMAP_SERVER and recompile or launch the octomap_color_server node
[ WARN] [1705566784.105559042]: Nothing to publish, octree is empty
按您推荐的知乎安装，还未解决，请问这个可以怎么解决呢？

silencht, 2024-01-19:
可能的解决方法已在之前的issues中体现，请查阅 :)

Maojiabao, 2024-03-24:
I have also encountered this problem. Have you resolved it？

silencht, 2024-03-25:
try.
https://blog.csdn.net/weixin_44694952/article/details/109003337
https://blog.csdn.net/qq_34131212/article/details/106803808#t12

closed 2024-04-09 by silencht