silencht/SG-SLAM #29: ncnn路径问题

nit-xhg, 2024-05-23:
[image]
已经修改过CMakeLists.txt文件，但是catkin_make 还是报错，如图所示，显示找不到ncnn的配置文件

silencht, 2024-05-24:
填写路径下有ncnnConfig.cmake或ncnn-config.cmake文件吗？请确认。

nit-xhg, 2024-05-24:
> 填写路径下有ncnnConfig.cmake或ncnn-config.cmake文件吗？请确认。

感谢你的回答。
有的，我通过在终端下直接设置这个ncnn的路径解决了，但是开启八叉树地图的命令显示
[image]
该launch文件中好像没有提到彩色节点

silencht, 2024-05-24:
该问题是个普遍的常见问题，请浏览参考之前issues，例如 https://github.com/silencht/SG-SLAM/issues/25

nit-xhg, 2024-05-24:
> 该问题是个普遍的常见问题，请浏览参考之前issues，例如 #25

感谢！

nit-xhg, 2024-05-24:
> 该问题是个普遍的常见问题，请浏览参考之前issues，例如 #25

八叉树地图的rviz文件参数需要修改吗，运行时报错
[image]

silencht, 2024-05-24:
可能是ROS版本问题，参考https://github.com/silencht/SG-SLAM/issues/20

nit-xhg, 2024-05-24:
> 可能是ROS版本问题，参考#20
运行orb_slam命令没有报错,并且能得到轨迹数据
我看之前有个问题里面的命令有个transform.launch,但是我现在并没有看到这个文件
[image]

silencht, 2024-05-24:
> 我看之前有个问题里面的命令有个transform.launch,但是我现在并没有看到这个文件

那个是ROS tf坐标系变换的文件，我觉得启动这么多命令太麻烦了，所以直接写进程序里面了。已经不需要它了。

nit-xhg, 2024-05-24:
都能运行起来，但是八叉树地图的launch文件依旧显示没有没有发布者

[image]

nit-xhg, 2024-05-24:
> > 我看之前有个问题里面的命令有个transform.launch,但是我现在并没有看到这个文件
> 
> 那个是ROS tf坐标系变换的文件，我觉得启动这么多命令太麻烦了，所以直接写进程序里面了。已经不需要它了。

并且有个这种警告
[image]

nit-xhg, 2024-05-28:
> > 都能运行起来，但是八叉树地图的launch文件依旧显示没有没有发布者
> > ![image](https://private-user-images.githubusercontent.com/90295680/333553560-fb320eb4-e6c3-4b87-b3b4-7a4438758265.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY4NzcwODQsIm5iZiI6MTcxNjg3Njc4NCwicGF0aCI6Ii85MDI5NTY4MC8zMzM1NTM1NjAtZmIzMjBlYjQtZTZjMy00Yjg3LWIzYjQtN2E0NDM4NzU4MjY1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI4VDA2MTMwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI2NTU2YWI2ZGQxMjc0Y2RmNWVlNGI3OTkzNjUxMmI0ODdmNWM0YjkzMzg1OTYyZDE5YmZkNzNjNTFmODc4MDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.LKNCXHdxFYY0STRgvu-ayCXfL5rETEp0JeTNLgwboLY)
> 
> 正常打开时也会提示一句，只要后续没有连续的[WARN]信息即可。 抱歉，下面那个我暂时没有头绪，建议通过关键字查询相关blog :(
十分感谢

closed 2024-07-10 by silencht