silencht/SG-SLAM #34: 没有tf变换数据

nit-xhg, 2024-10-01:
作者您好
我在使用自己录制的tum数据集时，发现特征点提取的过程中会有一段时间提取不到特征点，并且八叉树地图由于没有tf变换导致无法构建，God damnit, A is singular, this shouldn't happen。配置文件也换成了自己的配置文件，我想请问下为什么会出现这种问题。感谢您的解答。
[image]
[image]

silencht, 2024-10-01:
是不是前端跟踪丢失导致的？

image.png (view on web)
image.png (view on web)

nit-xhg, 2024-10-08:
> 是不是前端跟踪丢失导致的？
抱歉，这几天没看消息。应该是，但是我观察到前几帧的跟踪应该是正常的，想请问下为什么会出现这种情况呢

silencht, 2024-10-09:
视觉SLAM确实很容易丢失吧。丢失的原因有很多：
1. 设备算力不够，后端算的慢，拖累前端；
2. 场景高度动态，正确匹配特征过少；
3. 场景表示挑战大，特征少（相机视场角小导致的、场景本身导致的等等）、重复；
4. 场景光照挑战大，太亮太暗、突然亮突然暗（动态范围变化快）导致特征不行；
5. 设备轨迹太刁钻，大旋转、抖动、遮挡造成特征数据质量严重下降；
6. 设备本身数据质量低，传感器本身精度及局限性（比如深度相机扫描水、玻璃等物体）；
等等。

推测：前几帧正常，后面立即丢失，可能是前期地图建的太小，导致后面出现上述某个问题时，系统来不及纠正就挂了。

closed 2024-10-09 by silencht