silencht/SG-SLAM #35: 你好 我在尝试运行3D点云地图时 显示的效果是部分点云 这是为啥？

qinran123, 2024-10-30:
[image]

silencht, 2024-10-30:
可能的原因：
1. 场景中有动态对象（人）的话 => 剔除此部分点云 => 点云输入减少
2. 为计算效率考虑，程序每次只处理关键帧点云，如果朝向某场景得关键帧数量稀疏 => 点云输入减少
3. 场景中可能含有高反射物体（玻璃，水），剔除了低质量的深度信息 => 点云输入减少
4. 高动态场景可能导致短暂丢失 => 关键帧缺失 => 点云输入减少

qinran123, 2024-10-31:
我跑到 就是B站视频里的那个数据集

silencht, 2024-11-04:
`/SG_SLAM/Global_Point_Clouds` topic 显示全局点云地图
`/SG_SLAM/Local_Point_Clouds` topic 显示当前关键帧点云

closed 2024-11-04 by silencht

qinran123, 2024-11-10:
/SG_SLAM/Global_Point_Clouds topic 显示全局点云地图 在RVIZ界面选择Global_Point_Clouds 没有显示点云
[image]

qinran123, 2024-11-10:
[image]
在rviz界面 将Point_Clouds话题设为Global_Point_Clouds 没有显示点云

silencht, 2024-11-11:
请提供camera.yaml参数或者程序启动过程的log以便查看问题。