silencht/SG-SLAM #37: 训练自制数据集

blingblingDoris, 2024-11-01:
您好，本人slam小白想请问一下在训练自制数据集时，需要修改哪些文件呀，要注意什么问题，我用的是realsense d435i相机，非常感谢

silencht, 2024-11-04:
两个方法：
一个是按照TUM RGB-D Datasets的格式录制数据集，对应[可执行文件 rgbd_tum.cc](https://github.com/silencht/SG-SLAM/blob/main/src/sg-slam/Examples/rgbd_tum.cc)；
另一个推荐的方法是直接录制 ROS bag，输出 `RGB` 和 `Depth` 两路图像话题（Realsense 注意使用与 `RGB` 图对齐后的 `Depth` 图），后期可以根据 bag 的话题名称修改[可执行文件 ros_rgbd.cc](https://github.com/silencht/SG-SLAM/blob/main/src/sg-slam/Examples/ros_rgbd.cc)；

closed 2024-11-04 by silencht

blingblingDoris, 2024-11-04:
好的、非常感谢，我就是按照第二种方法改的，提示报错Usage: rosrun sg_slam_ros_rgbd path_to_vocabulary path_to_settings。另外、我还想问一下这个语义识别部分和语义对象度量信息输出部分的代码在哪里呀，非常感谢，我代码能力比较弱，麻烦了

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2024年11月04日 09:20 |
| 收件人 | silencht/SG-SLAM ***@***.***> |
| 抄送人 | blingblingDoris ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [silencht/SG-SLAM] 训练自制数据集 (Issue #37) |

两个方法：
一个是按照TUM RGB-D Datasets的格式录制数据集，对应可执行文件 rgbd_tum.cc；
另一个推荐的方法是直接录制 ROS bag，输出 RGB 和 Depth 两路图像话题（Realsense 注意使用与 RGB 图对齐后的 Depth 图），后期可以根据 bag 的话题名称修改可执行文件 ros_rgbd.cc；

silencht, 2024-11-06:
1. 报错是因为可执行文件用错了吧，需要按照提示的格式来执行 => 添加vocabulary路径。
2. 主要在[Detector2D.cc](https://github.com/silencht/SG-SLAM/blob/main/src/sg-slam/src/Detector2D.cc) 、[Detector3D.cc](https://github.com/silencht/SG-SLAM/blob/main/src/sg-slam/src/Detector3D.cc) 以及 [PointcloudMapping.cc](https://github.com/silencht/SG-SLAM/blob/main/src/sg-slam/src/PointcloudMapping.cc)

blingblingDoris, 2024-12-04:
您好、再次打扰一下、请问vtk和plc用的是什么版本的、现在我用的是vtk7.1和plc1.1.0，很多版本报错、非常感谢，期待您的回信

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2024年11月06日 14:31 |
| 收件人 | silencht/SG-SLAM ***@***.***> |
| 抄送人 | blingblingDoris ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [silencht/SG-SLAM] 训练自制数据集 (Issue #37) |

报错是因为可执行文件用错了吧，需要按照提示的格式来执行 => 添加vocabulary路径。
主要在Detector2D.cc 、Detector3D.cc 以及 PointcloudMapping.cc

blingblingDoris, 2024-12-12:
非常感谢、总是麻烦您、可以加一下WeChat帮忙解答一些问题吗（可以付费）总是问一些小问题、现在都有点不好意思问了，谢谢🙏

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2024年12月05日 23:36 |
| 收件人 | silencht/SG-SLAM ***@***.***> |
| 抄送人 | blingblingDoris ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [silencht/SG-SLAM] 训练自制数据集 (Issue #37) |

libvtk: 6.3
libpcl: 1.8.1