silencht/SG-SLAM #39: 无法运行目标检测网络

TNww, 2024-11-26:
已经安装了nvidia driver,vulkan，但是运行的时候无法生成语义地图

silencht, 2024-11-27:
请提供更详细的内容，包括启动终端日志。

TNww, 2024-11-27:
[ INFO] [1732693119.220434368]: rviz version 1.14.25
[ INFO] [1732693119.220466298]: compiled against Qt version 5.12.8
[ INFO] [1732693119.220480657]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1732693119.225516249]: Forcing OpenGl version 0.
[ INFO] [1732693119.566880106]: Stereo is NOT SUPPORTED
[ INFO] [1732693119.566908976]: OpenGL device: NVIDIA GeForce RTX 4060 Ti/PCIe/SSE2
[ INFO] [1732693119.566917786]: OpenGl version: 4.6 (GLSL 4.6).
[    0.000000] Linux version 5.15.0-126-generic (buildd@lcy02-amd64-052) (gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #136~20.04.1-Ubuntu SMP Thu Nov 14 16:38:05 UTC 2024 (Ubuntu 5.15.0-126.136~20.04.1-generic 5.15.167)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-126-generic root=UUID=17f37de0-12ae-4722-ba91-98b0d6f58ee7 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009e01fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009e02000-0x0000000009ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a000000-0x000000000a1fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000a200000-0x000000000a20ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000a210000-0x000000000affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000b000000-0x000000000b020fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000b021000-0x00000000a738efff] usable
[    0.000000] BIOS-e820: [mem 0x00000000a738f000-0x00000000ad38efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ad38f000-0x00000000ad47efff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000ad47f000-0x00000000af47efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000af47f000-0x00000000b85fefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b85ff000-0x00000000b9ff8fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b9ff9000-0x00000000b9ffbfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b9ffc000-0x00000000b9ffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ba000000-0x00000000bbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bd7f3000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f7000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000083defffff] usable
[    0.000000] BIOS-e820: [mem 0x000000083ef40000-0x00000008601fffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000fd00000000-0x000000ffffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0xa052d018-0xa0551c57] usable ==> usable
[    0.000000] e820: update [mem 0xa052d018-0xa0551c57] usable ==> usable
[    0.000000] e820: update [mem 0xa1ece018-0xa1ed8e57] usable ==> usable
[    0.000000] e820: update [mem 0xa1ece018-0xa1ed8e57] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000009e01fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000009e02000-0x0000000009ffffff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000a000000-0x000000000a1fffff] usable
[    0.000000] reserve setup_data: [mem 0x000000000a200000-0x000000000a20ffff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000000a210000-0x000000000affffff] usable
[    0.000000] reserve setup_data: [mem 0x000000000b000000-0x000000000b020fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000b021000-0x00000000a052d017] usable
[    0.000000] reserve setup_data: [mem 0x00000000a052d018-0x00000000a0551c57] usable
[    0.000000] reserve setup_data: [mem 0x00000000a0551c58-0x00000000a1ece017] usable
[    0.000000] reserve setup_data: [mem 0x00000000a1ece018-0x00000000a1ed8e57] usable
[    0.000000] reserve setup_data: [mem 0x00000000a1ed8e58-0x00000000a738efff] usable
[    0.000000] reserve setup_data: [mem 0x00000000a738f000-0x00000000ad38efff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ad38f000-0x00000000ad47efff] ACPI data
[    0.000000] reserve setup_data: [mem 0x00000000ad47f000-0x00000000af47efff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x00000000af47f000-0x00000000b85fefff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000b85ff000-0x00000000b9ff8fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000b9ff9000-0x00000000b9ffbfff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000b9ffc000-0x00000000b9ffffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000ba000000-0x00000000bbffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000bd7f3000-0x00000000bfffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000f7000000-0x00000000ffffffff] reserved
[... cut ...]
[ 2605.613528] usb 1-1: Device not responding to setup address.
[ 2605.857533] usb 1-1: Device not responding to setup address.
[ 2606.074818] usb 1-1: device not accepting address 29, error -71
[ 2606.278836] usb 1-1: new full-speed USB device number 30 using xhci_hcd
[ 2606.305556] usb 1-1: Device not responding to setup address.
[ 2606.553535] usb 1-1: Device not responding to setup address.
[ 2606.770828] usb 1-1: device not accepting address 30, error -71
[ 2606.780717] usb usb1-port1: unable to enumerate USB device

TNww, 2024-11-27:
我参考了之前的问题，把/map改为map，仍无语义地图

silencht, 2024-11-27:
(⊙o⊙)呃，其实我要的是SG-SLAM启动之后的终端输出信息，不是Linux系统启动日志。
另外还有语义地图的错误信息。

TNww, 2024-11-27:
实在不好意思qwq
roslaunch octomap.launch... logging to /home/ww/.ros/log/7db36c0c-acb8-11ef-99f6-1919f13d28ce/roslaunch-www-2111.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://www:46371/

SUMMARY
========

PARAMETERS
 * /octomap_server/colored_map: True
 * /octomap_server/filter_ground: False
 * /octomap_server/filter_speckles: False
 * /octomap_server/frame_id: map
 * /octomap_server/ground_filter/angle: 0.15
 * /octomap_server/ground_filter/distance: 0.05
 * /octomap_server/ground_filter/plane_distance: 0.05
 * /octomap_server/height_map: False
 * /octomap_server/latch: False
 * /octomap_server/occupancy_max_z: 1.5
 * /octomap_server/occupancy_min_z: -1.5
 * /octomap_server/pointcloud_max_z: 5.0
 * /octomap_server/pointcloud_min_z: -5.0
 * /octomap_server/resolution: 0.03
 * /octomap_server/sensor_model/hit: 0.7
 * /octomap_server/sensor_model/max: 0.97
 * /octomap_server/sensor_model/max_range: -1.0
 * /octomap_server/sensor_model/min: 0.12
 * /octomap_server/sensor_model/miss: 0.4
 * /rosdistro: noetic
 * /rosversion: 1.17.0

NODES
  /
    octomap_server (octomap_server/octomap_server_node)

ROS_MASTER_URI=http://localhost:11311

process[octomap_server-1]: started with pid [2125]
[ WARN] [1732709399.447895545]: Nothing to publish, octree is empty
[ INFO] [1732709405.942085831]: rviz version 1.14.25
[ INFO] [1732709405.942121891]: compiled against Qt version 5.12.8
[ INFO] [1732709405.942129982]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1732709405.949651460]: Forcing OpenGl version 0.
[ INFO] [1732709406.297615857]: Stereo is NOT SUPPORTED
[ INFO] [1732709406.297644427]: OpenGL device: NVIDIA GeForce RTX 4060 Ti/PCIe/SSE2
[ INFO] [1732709406.297653507]: OpenGl version: 4.6 (GLSL 4.6).

silencht, 2024-11-28:
有SG-SLAM启动过后
```yaml
PointCloudMapping.is_map_construction_consider_dynamic: 1

PointCloudMapping.camera_valid_depth_Min: 0.5
PointCloudMapping.camera_valid_depth_Max: 5.0

PointCloudMapping.is_octo_semantic_map_construction: 1
PointCloudMapping.Sor_Local_MeanK: 50
PointCloudMapping.Sor_Local_StddevMulThresh: 3.0
PointCloudMapping.Voxel_Local_LeafSize: 0.01

PointCloudMapping.is_global_pc_reconstruction: 0
PointCloudMapping.Sor_Global_MeanK: 50
PointCloudMapping.Sor_Global_StddevMulThresh: 3.0
PointCloudMapping.Voxel_Global_LeafSize: 0.01

Detector3D.Sor_MeanK: 50
Detector3D.Sor_StddevMulThresh: 1.0
Detector3D.Voxel_LeafSize: 0.01
Detector3D.EuclideanClusterTolerance: 0.02
Detector3D.EuclideanClusterMinSize: 1000
Detector3D.EuclideanClusterMaxSize: 30000
Detector3D.DetectSimilarCompareRatio: 0.1
Detector3D.global_pc_update_kf_threshold: 25

Detector2D.detection_confidence_threshold: 0.90
Detector2D.dynamic_detection_confidence_threshold: 0.01
```
日志信息吗？
运行的自己的摄像头还是数据集，目标检测的OpenCV窗口显示boungdingbox吗？

silencht, 2024-11-29:
`find_blob_index_by_name input failed` 说明推理出现了错误，ncnn库的配置存在问题。建议先跑通ncnn的官方examples或benchmarks。

TNww, 2024-11-30:
[image]

silencht, 2024-11-30:
请提供右下角第四个窗口的完整终端输出

TNww, 2024-11-30:
[image]

TNww, 2024-11-30:
fopen ./Thirdparty/ncnn_model/mobilenetv3_ssdlite_voc.param failed
fopen ./Thirdparty/ncnn_model/mobilenetv3_ssdlite_voc.bin failed

感谢作者，我好像发现问题了，或许您有解决的建议嘛，检查过对应路径下有相应文件

TNww, 2024-11-30:
ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

is map construction consider dynamic: 1
is global pointcloud reconstruction: 0
is octomap semanticObject map construction: 1
camera_valid_depth_Min is 0.5
camera_valid_depth_Max is 5
Sor_Local_MeanK is 50
Sor_Local_StddevMulThresh is 3
Voxel_Local_LeafSize is 0.01
Sor_Global_MeanK is 50
Sor_Global_StddevMulThresh is 3
Voxel_Global_LeafSize is 0.01
Detector3D_Sor_MeanK is 50
Detector3D_Sor_StddevMulThresh is 1
Detector3D_Voxel_LeafSize is 0.01
Detect3D_EuclideanClusterTolerance is 0.02
Detect3D_EuclideanClusterMinSize is 1000
Detect3D_EuclideanClusterMaxSize is 30000
Detect3D_DetectSimilarCompareRatio is 0.1
global_pc_update_kf_threshold is 25
Start PointCloudMapping Viewer...

Camera Parameters: 
- fx: 535.4
- fy: 539.2
- cx: 320.1
- cy: 247.6
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 2.98842

detection_confidence_threshold is 0.9

dynamic_detection_confidence_threshold is 0.01
fopen ./Thirdparty/ncnn_model/mobilenetv3_ssdlite_voc.param failed
fopen ./Thirdparty/ncnn_model/mobilenetv3_ssdlite_voc.bin failed
Detector2D Thread start ...

-------
Start processing sequence ...
Images in the sequence: 827

find_blob_index_by_name input failed
Try
find_blob_index_by_name detection_out failed
Try
New map created with 718 points
Gtk-Message: 15:13:39.460: Failed to load module "canberra-gtk-module"
find_blob_index_by_name input failed
Try
find_blob_index_by_name detection_out failed
Try
find_blob_index_by_name input failed
Try
find_blob_index_by_name detection_out failed
Try
find_blob_index_by_name input failed
Try
find_blob_index_by_name detection_out failed
Try
find_blob_index_by_name input failed
Try
find_blob_index_by_name detection_out failed
Try
find_blob_index_by_name input failed
Try
find_blob_index_by_name detection_out failed
Try
find_blob_index_by_name input failed
Try
find_blob_index_by_name detection_out failed
Try
find_blob_index_by_name input failed
Try
^CMakeCurrent: Not available with non-pangolin window.
-------

median tracking time: 0
mean tracking time: 0.00118231

Saving camera trajectory to CameraTrajectory.txt ...

trajectory saved!

Saving keyframe trajectory to KeyFrameTrajectory.txt ...

trajectory saved!

TNww, 2024-12-02:
感谢！解决了！

closed 2024-12-02 by silencht