silencht/SG-SLAM #40: catkin_make --pkg sg-slam       There was an error at runtime？

Sun173558, 2024-12-01:
/usr/include/c++/11/bits/stl_map.h: In instantiation of ‘class std::map<ORB_SLAM2::KeyFrame*, g2o::Sim3, std::less<ORB_SLAM2::KeyFrame*>, Eigen::aligned_allocator<std::pair<const ORB_SLAM2::KeyFrame*, g2o::Sim3> > >’:
/home/hri/wenke/xiaos/1/SG-SLAM/src/sg-slam/src/Optimizer.cc:818:37:   required from here
/usr/include/c++/11/bits/stl_map.h:123:71: error: static assertion failed: std::map must have the same value_type as its allocator
  123 |       static_assert(is_same<typename _Alloc::value_type, value_type>::value,
      |                                                                       ^~~~~

silencht, 2024-12-04:
https://zhuanlan.zhihu.com/p/218019316

#12

Sun173558, 2024-12-04:
Thank you very much, I've solved this problem, but when I ran workling.sh that script, I kept resetting when I built the map, and finally it showed a segregal error and crashed, what is the reason for this, and that map, all the /map I modified the map, or there is an error display, if you can reply, I would be grateful

---- Replied Message ----
| From | ***@***.***> |
| Date | 12/04/2024 10:09 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [silencht/SG-SLAM] catkin_make --pkg sg-slam       There was an error at runtime？ (Issue #40) |

https://zhuanlan.zhihu.com/p/218019316

#12

silencht, 2024-12-04:
please paste your (running walking.sh script's) terminal output below this issue.

Sun173558, 2024-12-04:
It's like this, the running terminal keeps resetting and then the second terminal shows an empty octree.It's not continuous when loading images, it's like this

---- Replied Message ----
| From | ***@***.***> |
| Date | 12/04/2024 10:34 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [silencht/SG-SLAM] catkin_make --pkg sg-slam       There was an error at runtime？ (Issue #40) |

Reopened #40.

Sun173558, 2024-12-04:
Breakpoint debugging shows this

---- Replied Message ----
| From | ***@***.***> |
| Date | 12/04/2024 10:34 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [silencht/SG-SLAM] catkin_make --pkg sg-slam       There was an error at runtime？ (Issue #40) |

Reopened #40.

silencht, 2024-12-04:
sorry, but i can't see your image?

closed 2025-02-20 by silencht