silencht/SG-SLAM #42: 真实数据集录制

Ws1234567879789, 2024-12-20:


silencht, 2024-12-23:
请提供运行视频以及终端日志

Ws1234567879789, 2024-12-23:
请将你的邮箱给我下，谢谢

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2024年12月23日 09:24 |
| 收件人 | silencht/SG-SLAM ***@***.***> |
| 抄送人 | Ws1234567879789 ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [silencht/SG-SLAM] 真实数据集录制 (Issue #42) |

请提供运行视频以及终端日志

silencht, 2024-12-23:
***@***.***

 
 | 发件人 | ***@***.***&gt; |
 | 发送日期 | 2024年12月23日 09:24 |
 | 收件人 | silencht/SG-SLAM ***@***.***&gt; |
 | 抄送人 | Ws1234567879789 ***@***.***&gt;,
 Author ***@***.***&gt; |

Ws1234567879789, 2024-12-23:
你好，你的邮箱被官方隐藏了，全是*，您看可不可以发图片

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2024年12月23日 11:51 |
| 收件人 | silencht/SG-SLAM ***@***.***> |
| 抄送人 | Ws1234567879789 ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [silencht/SG-SLAM] 真实数据集录制 (Issue #42) |

***@***.***

silencht, 2024-12-23:
你好，我 GitHub 主页里面有我的邮箱。

 
 | 发送日期 | 2024年12月23日 11:51 |
 
 
 
 
 
 
 
 
 
 | 发件人 | ***@***.***&amp;gt; |
 | 发送日期 | 2024年12月23日 09:24 |
 | 收件人 | silencht/SG-SLAM ***@***.***&amp;gt; |
 | 抄送人 | Ws1234567879789 ***@***.***&amp;gt;,
 Author ***@***.***&amp;gt; |

Ws1234567879789, 2024-12-23:
我看了一下，没看到你的邮箱

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2024年12月23日 12:38 |
| 收件人 | silencht/SG-SLAM ***@***.***> |
| 抄送人 | Ws1234567879789 ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [silencht/SG-SLAM] 真实数据集录制 (Issue #42) |

你好，我 GitHub 主页里面有我的邮箱。

silencht, 2024-12-24:
silencht[at]qq[dot]com